sqfmi/beepy-hardware #6: Consolidate expansion I/O into single stackable header in v2

57Bravo, 2023-06-09:
I like that you broke the extra I/O out for expansion. Mechanically though, I'd like to see it consolidated into single stack able header. I'd also switch from through hole to SMD like J45. You could probably route of all the expansion down just below the RPI header. 
 
[image]

sockbot, 2023-06-10:
What about breaking out the gpio in the same layout as on the RPi itself? Then there would be compatibility with the entire ecosystem of HATs and accessories.

57Bravo, 2023-06-11:
Possibly. PCB real estate is pretty tight though. If you put another full 0.1" pitch header in there, that would be significant. Anything that stacks needs to fit in a 10mm ish depth, or the thing goes from handheld to laptop. 

I do like common interfaces though. SFEs micromod / qwiic are some good examples.