sqfmi/beepy-hardware #8: Create a LoRa transceiver add-on in v2

57Bravo, 2023-06-09:
I'm not sure how you'd package it, but it would be neat to integrate a LoRa transceiver in the next version of the board. You could use it to send and receive text like messages using LoRa gateways. 

Use cases:
- Burst based text messaging
- One way broadcast paging 
- SOS (that's a stretch without a GPS too, but maybe a coarse location could be calculated using multiple gateways w/o GPS.)

Aemilivs, 2023-06-09:
Imagine rendering global mobile communication infrastructure obsolete by introducing Lora circuit and application utilizing public keys for it in beepberry lmao

vitasam, 2023-06-09:
LoRa is a really good idea

57Bravo, 2023-06-09:
I'm thinking something like what they do with atak-civ. 

https://meshtastic.org/docs/software/community/community-atak

I mean, no gui. Just the concept of the LoRa bridge for messaging.

vitasam, 2023-06-10:
Definitely no GUI!

And with mesh?

https://hackaday.com/2020/02/26/lora-mesh-network-with-off-the-shelf-hardware/

57Bravo, 2023-06-11:
Someone on the uConsole forums posted this. It's like beepberry with an mcu and onboard LoRa. https://www.lilygo.cc/products/t-deck

57Bravo, 2023-06-13:
Here's another LoRa/berry device. 

https://hackaday.com/2022/05/25/long-distance-text-communication-with-lora/

https://github.com/BigCorvus/LORA-QWERTY-Communicator