sqfmi/beepy-hardware #17: V1 Case needs a power slide button

57Bravo, 2023-07-14:
When doing development on the keyboard, I'm constantly sliding the power button on / off with the V1 case. I have to use a small screwdriver because the switch is tiny. 

It would be nice it the V1 case had a small slide button so you could use your finger instead.

57Bravo, 2023-10-02:
FW updates now allow upper right button to gracefully power and shutdown rpi without sliding switch.

closed 2023-10-02 by 57Bravo

prokrypt, 2023-10-02:
I use the corner of a usb-c plug to toggle the switch.

57Bravo, 2023-10-02:
If you update the keyboard firmware and associated kernel module, the upper right hand button allows for digital power up/down. They also exposed the led to sysfs, so you can control it without unloading the keyboard kernel module to hit I2C directly.

prokrypt, 2023-10-02:
My switch does soft shutdown too, somehow...
[image]

57Bravo, 2023-10-02:
Mod city ha. Here's mine https://github.com/57Bravo/BeepSensors