sqfmi/beepy-hardware #18: Add a clear qwerty key mapping to docs

57Bravo, 2023-07-14:
The qwerty keyboard doesn't clearly show alternate symbols like: [,],<,>,|, tab, pg up/down etc. It also doesn't show common modifiers like ctrl-c.

Add some documentation that talks about the key map for the default firmware.

It took me a couple of hours of fiddling to figure out the map so I could use tmux on the device.

57Bravo, 2023-07-18:
# bbqX0kbd_driver/source/mod_src/bbq20kbd_pmod_codes.h

```
/*
 *                                      BBQ20KBD PMOD KEYBOARD LAYOUT
 *
 *      +------+-----+----+----+----+----+----+-----+-----+-------+
 *      |      |          |BR     ↑TPY-       |           |       |
 *      | Ctrl |   PgDn   |←TPX- BL(HOME)TPX+→|   PgUp    | MENU  |
 *      |      |          |       ↓TPY+       |           |       |
 *      +------+-----+----+----+----+----+----+-----+-----+-------+
 *      |                                                         |
 *      +------+-----+----+----+----+----+----+-----+-----+-------+
 *      |#     |1    |2   |3   |(   |   )|_   |    -|    +|      @|
 *      |  Q   |  W  | E  | R  | T  |  Y |  U |  I  |  O  |   P   |
 *      |      |     |PgDn|PgUp|   \|UP  |^   |=    |{    |}      |
 *      +------+-----+----+----+----+----+----+-----+-----+-------+
 *      |*     |4    |5   |6   |/   |   :|;   |    '|    "|    ESC|
 *      |  A   |  S  | D  | F  | G  |  H |  J |  K  |  L  |  BKSP |
 *      |     ?|     |   [|   ]|LEFT|HOME|RGHT|V+   |V-   |DLT    |
 *      +------+-----+----+----+----+----+----+-----+-----+-------+
 *      |      |7    |8   |9   |?   |   !|,   |    .|    `|       |
 *      |LFTALT|  Z  | X  | C  | V  |  B |  N |  M  |  $  | ENTER |
 *      |      |   K+|  K-|   °|   <|DOWN|>   |MENU |Vx   |       |
 *      +------+-----+----+----+----+----+----+-----+-----+-------+
 *      |            |0   |TAB                |     |             |
 *      | LEFT_SHIFT | ~  |       SPACE       |RTALT| RIGHT_SHIFT |
 *      |            |  Kx|                  &|     |             |
 *      +------------+----+-------------------+-----+-------------+
 *
 * Notes:
 * Most Important to know:
 * To present a character as mentioned above a key in the above layout, use LFTALT + Key.
 * To present a character as mentioned below a key in the above layout, use RTALT + Key.
 *
 * Changes from arturo182's layout, if you are used to that, it's good to know these changes.
 * 1. Sym Key (Scancode 0x1D) on Keyboard is mapped to RTALT.
 * 3. The external buttons on the Keyboard Featherwing have been changed as below:
 *__3.1 Leftmost Button changed from Menu Key to Left Ctrl Key.
 *__3.2 Innet Left Button changed from Left Ctrl Key to Page Up Key.
 *__3.3 Inner Right Button changed from Back Key to Page Down Key.
 *__3.4 Outer Right Button changed from Left Shift Key to Menu Key.
 *__3.5 Center Key of 5-way Button changed from Enter Key to Home Key.
 */

```

sqfmi, 2023-09-08:
Docs updated with latest PR: https://beepy.sqfmi.com/docs/firmware/keyboard#symbol-key-map

closed 2023-09-08 by 57Bravo