sqfmi/beepy-hardware #19: Add a piezo element for audio in Rev A HW

57Bravo, 2023-08-05:
I was able to use a piezo element to get my beepberry to speak using espeak. It would be nice if the next rev of HW had a buzzer/piezo for indications and audio.

A little tinny, but it's usable.

[image]

https://github.com/sqfmi/beepy-hardware/assets/17078019/80972862-f4a9-4f71-973d-3ff777216369

sockbot, 2023-08-05:
Could you upload a schematic and a BOM to this request so that I could look into reproducing it? Pictures of hardware install would be great too.

57Bravo, 2023-08-06:
Sure thing. I pushed the repo here. 

https://github.com/57Bravo/BeepSensors